osrf/gazebo_models #43: McMillan Airfield GPS origin is not correct

Jaeyoung-Lim, 2020-05-18:
When comparing to the map, the origin of the GPS coordinates are not correct and seems to be off a few hundred meters.

I tried fixing this comparing on google maps, but changing
https://github.com/PX4/sitl_gazebo/blob/12da9e6104aaf72a0c52bf802644310f0c5b1ee1/worlds/mcmillan_airfield.world#L18-L23

Didn't have an effect on setting the gps origins. It seems like `https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_gps_plugin.cpp#L51` is not able to be modified.

@ahcorde Is there somewhere else that the world origins are set?

Cross posted from https://github.com/PX4/sitl_gazebo/issues/496

issouker97, 2020-07-12:
Hi , unfortunately i don't have any reply for your question , in the other hand i want to add person_walking  i clone it but i don't know how to launch it in gazebo , actually i need the command to launch it , i would be grateful if you could help me

Jaeyoung-Lim, 2020-07-25:
@issouker97 I belive your issue is not related to your issue. Please don't hijack the topic of this thread

Jaeyoung-Lim, 2020-08-10:
@ahcorde Could you shed any light on this issue?

ahcorde, 2020-08-13:
The tags that you mentioned are being readed in the Gazebo code ( in the [world class](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/physics/World.cc#L316) ).

In the plugin you need to setup some [env vars](https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_gps_plugin.cpp#L94):

```
 // Use environment variables if set for home position.
  const char *env_lat = std::getenv("PX4_HOME_LAT");
  const char *env_lon = std::getenv("PX4_HOME_LON");
  const char *env_alt = std::getenv("PX4_HOME_ALT");
```

Jaeyoung-Lim, 2020-08-13:
@ahcorde Without the environment variables, it reads from the worldfile with https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_gps_plugin.cpp#L105, However even if I change the spherical coordinates in https://github.com/PX4/sitl_gazebo/blob/master/worlds/mcmillan_airfield.world#L20 the returned value doesn't change

Jaeyoung-Lim, 2020-11-17:
@ahcorde I can confirm that when reading the spherical coordinates from the world object, it returns the origin set from the `mcmillan_airfield.tif` file and not the one that is defined in the `.world` file

Jaeyoung-Lim, 2020-11-17:
@ahcorde It seems like gazebo uses the origin of the DEM rather than the spherical coordinates set by the world file if I understand correctly the following

https://github.com/osrf/gazebo/blob/063d7386ed5499c2883de48e19ff5fb9dec01dff/gazebo/physics/HeightmapShape.cc#L109-L131